xhikarishii/openwrt-ipq #1: Bridge Mode

MiG-41, 2024-07-20:
### Describe the bug

We trying to use bridge mode ( bridge wwan0 to lan4 or wan i.e.)  , in witch quactel-CM can do it : https://eko.one.pl/forum/viewtopic.php?id=24045
In your build i stack in adding wwan0 to bridge , it always said ,that it is busy. 
Tried to delete hotplug for wwan , also force remove ModemManager , but all the time is busy after 
`root@HikariWrt:~# rmmod qmi_wwan_q
root@HikariWrt:~# insmod qmi_wwan_q bridge_mode=1`

Can you check in free time is it possible ?

### OpenWrt version

v1.1.0-Beta

### OpenWrt release

v1.1.0-Beta

### OpenWrt target/subtarget

v1.1.0-Beta

### Device

aw1000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

xhikarishii, 2024-07-26:
Hi @MiG-41, will look into it, will also look into disabling ModemManager on the next build as yes it causes issues on other protocols using the module.

xhikarishii, 2024-07-30:
@MiG-41  changing the label to `enhancement` since this is not really one but a feature request :)

xhikarishii, 2024-07-30:
Bridge mode is now enabled for `qmi_wwan_q` and will be included on the next release, closing this for now :)

MiG-41, 2024-07-30:
Did you test it ? Becouse issue was also , that it was not possible to add wwan0 to bridge (even with this option).

xhikarishii, 2024-07-30:
Yup testing it right now, but still experimenting on different configurations. 

I'll add default configuration along with the release, we are also testing `qmap-multiplexing` which we plan to include on the next release

MiG-41, 2024-07-30:
Multiplexing is as i belive same connection to different ANP's ? Not sure ,if this can be used ( Maybe one provier uses different APN for IPv4 and different for IPv6 ,but don't use hisservices). 
But bridge i looking forvard to test it :)

xhikarishii, 2024-07-31:
Reopening this ticket, as I experienced frequent Kernel Panics when the bridge is enabled. Will look further on this one.

MiG-41, 2024-07-31:
Can you give this release to test ? ( I understand that it can be not stable , but at least connecting , we were not able to recive a connection at all...)

xhikarishii, 2024-08-05:
Hi @MiG-41, it's not on a really usable state right now, the kernel panic will cause a bootloop for some reason. Will hold this tweak for now until we have resolve it. 

But for now I cannot promise anything as this is a complex issue and quectel support is really hard to pursue.

XT-Martinez, 2024-08-19:
@MiG-41 I saw a user manual for quectel-cm and it seems that macro `QUECTEL_BRIDGE_MODE` must be enabled in the driver source code for bridge mode to work.
https://sixfab.com/wp-content/uploads/2024/01/Quectel_QConnectManager_Linux_User_Guide_V1.0.pdf

MiG-41, 2024-08-19:
Yes , Cezary made a version with this option copmpiled, on his Teltonika RUT950 with EC25 everything went ok ,it was possible to add wwan0 to bridge , device received IP , but there where no transmission at all.
In my case with AW1000 i couldn't add wwan0 to bridge.

XT-Martinez, 2024-08-19:
Since qmap_mode is set to 1 on Hikari's build, perhaps it should be wwan0_1 instead of wwan0?
Can you also try `insmod qmi_wwan_q qmap_mode=0` and try again the bridge again with wwan0?

xhikarishii, 2024-09-26:
We just finished another release but I still can't manage to get bridge mode to work, I'll look further into this.

xhikarishii, 2024-09-26:
@MiG-41 Hi, I have stumbled upon the eko forums and seen that you are testing my builds there. If you can help me with identifying where the bottlenecks of the device that would really help. 

Currently I've been maxing out only around 700-800mbps on the device, and reception is really bad on my area (But on process on moving to the city) so hopefully I can get better overview there.

MiG-41, 2024-09-27:
Unfortunatly i dont have AW1000 anymore , so i can't test your newer 1.2 relese.

On 1.1 max. what i could get was 600Mbps... I wondering ,if that is max. on such connection B1+B7+n78 (n78 with 100Mhz bandwitch) , since on other devices and phones i get in the same place near tower 900Mbps...

xhikarishii, 2024-09-27:
> Unfortunatly i dont have AW1000 anymore , so i can't test your newer 1.2 relese.
> 
> On 1.1 max. what i could get was 600Mbps... I wondering ,if that is max. on such connection B1+B7+n78 (n78 with 100Mhz bandwitch) , since on other devices and phones i get in the same place near tower 900Mbps...

Would be really nice if anyone can share cell info and diagnostics. But will continue to look into the bottlenecks :)

xhikarishii, 2025-06-24:
Closing, unfortunately, bridge mode is not supported by AW1000's SOC

closed 2025-06-24 by xhikarishii