xhikarishii/openwrt-ipq #2: Quectel Cellular, QMI Cellular, ModemManager Protocols

atan2005, 2024-07-25:
### Describe the bug

I have to perform steps below for the modem to register and obtain ip address with my ISP (Optus and Telstra):-

1. When initially checking the Modem - Details - 3ginfo-lite screen, all entries were blank.
2. Under the Network - Interfaces - 5g, I have to first change from Protocol: _Quectel Cellular_ to _QMI Cellular_, and check the 3ginfo-lite screen again, this time some entries appears as it has registered with my ISP, however the Connection statistics: Waiting for connection data...
3. Repeating step 2 by reversing the Protocol: back to _Quectel Cellular_, this time Connection statistics is showing data!! I have left APN: as _internet_ for Optus and as _telstra.internet_ for Telstra, PDP Type: as IPv4/IPv6 and remember to save and restart the interface during each step.
4. If I reboot the modem, I need to repeat step 1 to 3 for internet connection, so something is broken here.
5. If I add a protocol under Network Interface such as _ModemManager_ in addition of existing _Quectel Cellular_ interface, this setup will survive a reboot or power-off/on and will connect automatically. ModemManager interface will show _Error: Unknown error (NEGOTIATION_FAILED)__
6. If I only run the _ModemManager_ on its own, it will register and connect to my ISP however strange default gateway ip address 10.64.64.64 is returned. The router can connect to internet resources, but NAT client pc cannot and the default gateway ip address cannot be pinged at all.

Please see attached pictures of each scenarios (I have to change from your default Argon theme that was causing the "_scrolling capture_" screen issue to using either Bootstrap, BootstrapDark or BootstrapLight theme).

1. When only running _QMI Cellular_ protocol.
[image]
[image]
[image]

2. When running _Quectel Cellular_ protocol in parallel with _ModemManager_ protocol.
[image]
[image]
[image]

3. When only running _ModemManager_ protocol.
[image]
[image]
[image]

### OpenWrt version

r26991-3ac5a2b54f

### OpenWrt release

v1.1.0b

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Arcadyan AW1000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MiG-41, 2024-07-26:
Is it not enough just to do after some time of  boot/reset 
/etc/init.d/network restart
?

xhikarishii, 2024-07-26:
@atan2005 can you show me the configuration tab of your 3ginfo lite? See if it's set to `/dev/ttyUSB2` as the diag port.

atan2005, 2024-07-26:
Under Modem - Information about 3G/4G/5G connection - Configuration tab, they were already set as Interface: `wan` and IP address / port for communication with the modem: `/dev/ttyUSB2`. This is the same under Modem - AT Commands - Configuration - Port for communication with the modem: `/dev/ttyUSB2` for when testing various AT commands.

[image]

Under the Network - Interfaces when using the Quectel and QMI protocols, can only set Modem device: `/dev/cdc-wdm0` and when using ModemManager can only set Modem device: `Quectel - RG500Q-EA`.

MiG-41, 2024-07-26:
Why you are adding mm interface  , it is not needed, v1.1.0-beta should connect with default settings . 
Only thing with could be to correct is:
- change the modem type to QMI ( but this is only once , and setting is permanent in modem , also when you flash other fw. Only way to change it is manually switch to other type via AT command ).
- in 5g interface  APN (change to yours ,if "internet" is not the one on your operator)  and PDP Type latest . After that turn off/on ( best from power) , and finally PIN ( but i belive no one use it). And there should be connection.

If it is not connecting , please do after system boot: /etc/init.d/network restart , or ifup wan ,and see what happen.

atan2005, 2024-07-27:
I have already tried to softboot and power off/on many times during each step and/or each scenario before coming to the conclusion. Please refer to my original post: step 2 and scenario 1, when running `QMI Cellular` protocol on its own for updates.

`AT+QCFG="usbnet"`
+QCFG: "usbnet",0
OK

`AT+CGDCONT?`
+CGDCONT: 1,"IPV4V6","telstra.internet","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0,,,,,,,,,"",,,,0
+CGDCONT: 2,"IP","telstra.internet","0.0.0.0",0,0,0,0,,,,,,,,,"",,,,0
+CGDCONT: 3,"IPV6","telstra.internet","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0,,,,,,,,,"",,,,0
OK

`AT+CGCONTRDP`
+CGCONTRDP: 1,5,telstra.internet,10.102.225.113,32.1.128.4.20.128.63.27.180.135.251.253.123.243.131.6, 254.128.0.0.0.0.0.0.0.0.0.0.0.0.0.1,10.3.56.162 32.1.128.4.44.66.11.22.0.0.0.0.0.0.0.2,10.3.8.2 32.1.128.4.45.67.12.0.0.0.0.0.0.0.0.1
OK

`AT+QNWPREFCFG="mode_pref"`
+QNWPREFCFG: "mode_pref",AUTO
OK

`AT+QNWPREFCFG="lte_band"`
+QNWPREFCFG: "lte_band",1:2:3:4:5:6:7:8:9:10:11:12:13:14:17:18:19:20:21:23:24:25:26:27:28:29:30:31:32:33:34:35:36:37:38:39:40:41:42:43:46:47:48:49:66:67:68:71:250:252:255
OK

`AT+QNWPREFCFG="nsa_nr5g_band"`
+QNWPREFCFG: "nsa_nr5g_band",1:2:3:5:7:8:12:14:20:25:28:34:38:39:40:41:48:50:51:65:66:70:71:74:75:76:77:78:79:80:81:82:83:84:85:86:257:258:259:260:261
OK

`AT+QNWPREFCFG="nr5g_band"`
+QNWPREFCFG: "nr5g_band",1:2:3:5:7:8:12:14:20:25:28:34:38:39:40:41:48:50:51:65:66:70:71:74:75:76:77:78:79:80:81:82:83:84:85:86:257:258:259:260:261
OK

`AT+QMBNCFG="Select"`
+QMBNCFG: "Select",Telstra_Australia_Commercial
OK

`AT+QMBNCFG="AutoSel"`
+QMBNCFG: "AutoSel",1
OK

`AT+QMBNCFG="List"`
+QMBNCFG: "List",0,1,1,"Telstra_Australia_Commercial",0x0A010F00,202111231
+QMBNCFG: "List",1,0,0,"Optus_Australia_Commercial",0x0A014400,202101151
+QMBNCFG: "List",2,0,0,"Germany-VoLTE-Vodafone",0x0A010449,202101151
+QMBNCFG: "List",3,0,0,"ChunghwaTel_Taiwan_Commercial",0x0A014F00,202104021
+QMBNCFG: "List",4,0,0,"Spark_Commercial",0x0A01FE09,202101151
+QMBNCFG: "List",6,0,0,"TIM_Italy_Commercial",0x0A012B00,202101151
+QMBNCFG: "List",7,0,0,"France-Commercial-Orange",0x0A010B21,202101151
+QMBNCFG: "List",8,0,0,"Commercial-DT-VOLTE",0x0A011F1F,202108201
+QMBNCFG: "List",9,0,0,"UK-VoLTE-Vodafone",0x0A010426,202101151
+QMBNCFG: "List",10,0,0,"Commercial-EE",0x0A01220B,202101151
+QMBNCFG: "List",11,0,0,"Commercial-LGU",0x0A012608,202101151
+QMBNCFG: "List",12,0,0,"Commercial-KT",0x0A01280B,202101151
+QMBNCFG: "List",13,0,0,"Commercial-SKT",0x0A01270A,202101151
+QMBNCFG: "List",14,0,0,"Commercial-Reliance",0x0A011B0C,202101151
+QMBNCFG: "List",15,0,0,"Commercial-SBM",0x0A011C0B,202101151
+QMBNCFG: "List",16,0,0,"Commercial-KDDI",0x0A010709,202101151
+QMBNCFG: "List",17,0,0,"Commercial-DCM",0x0A010D0D,202101151
+QMBNCFG: "List",19,0,0,"VoLTE_OPNMKT_CT",0x0A0113E0,202101151
+QMBNCFG: "List",20,0,0,"Volte_OpenMkt-Commercial-CMCC",0x0A012010,202101151
+QMBNCFG: "List",21,0,0,"ROW_Commercial",0x0A010809,202201041
OK

MiG-41, 2024-07-27:
I sugggest to deactivate MBN files (use a manual configuration ) , so

AT+QMBNCFG="Deactivate"
AT+QMBNCFG="AutoSel",0
And then check , if there are no "x,1,1" at any files ( Now "Telstra_Australia_Commercial" has it).

atan2005, 2024-07-28:
`AT+QMBNCFG="Deactivate"`

`AT+QMBNCFG="AutoSel",0`
OK

`AT+QMBNCFG="List"`
+QMBNCFG: "List",0,0,0,"Telstra_Australia_Commercial",0x0A010F00,202111231
+QMBNCFG: "List",1,0,0,"Optus_Australia_Commercial",0x0A014400,202101151
+QMBNCFG: "List",2,0,0,"Germany-VoLTE-Vodafone",0x0A010449,202101151
+QMBNCFG: "List",3,0,0,"ChunghwaTel_Taiwan_Commercial",0x0A014F00,202104021
+QMBNCFG: "List",4,0,0,"Spark_Commercial",0x0A01FE09,202101151
+QMBNCFG: "List",5,0,0,"Telia_Sweden",0x0A012400,202101151
+QMBNCFG: "List",6,0,0,"TIM_Italy_Commercial",0x0A012B00,202101151
+QMBNCFG: "List",7,0,0,"France-Commercial-Orange",0x0A010B21,202101151
+QMBNCFG: "List",8,0,0,"Commercial-DT-VOLTE",0x0A011F1F,202108201
+QMBNCFG: "List",9,0,0,"UK-VoLTE-Vodafone",0x0A010426,202101151
+QMBNCFG: "List",10,0,0,"Commercial-EE",0x0A01220B,202101151
+QMBNCFG: "List",11,0,0,"Commercial-LGU",0x0A012608,202101151
+QMBNCFG: "List",12,0,0,"Commercial-KT",0x0A01280B,202101151
+QMBNCFG: "List",13,0,0,"Commercial-SKT",0x0A01270A,202101151
+QMBNCFG: "List",14,0,0,"Commercial-Reliance",0x0A011B0C,202101151
+QMBNCFG: "List",15,0,0,"Commercial-SBM",0x0A011C0B,202101151
+QMBNCFG: "List",16,0,0,"Commercial-KDDI",0x0A010709,202101151
+QMBNCFG: "List",17,0,0,"Commercial-DCM",0x0A010D0D,202101151
+QMBNCFG: "List",18,0,0,"VoLTE-CU",0x0A011561,202101151
+QMBNCFG: "List",19,0,0,"VoLTE_OPNMKT_CT",0x0A0113E0,202101151
+QMBNCFG: "List",20,0,0,"Volte_OpenMkt-Commercial-CMCC",0x0A012010,202101151
+QMBNCFG: "List",21,0,0,"ROW_Commercial",0x0A010809,202201041
OK

MiG-41, 2024-07-28:
And now how looks APN?
AT+CGDCONT?
AT+CGCONTRDP
Is it ok now ?

atan2005, 2024-07-29:
`AT+CGDCONT?`
+CGDCONT: 1,"IPV4V6","telstra.internet","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0,,,,,,,,,"",,,,0
OK

`AT+CGCONTRDP`
OK

When only running QMI Cellular protocol.
[image]
[image]
[image]

xhikarishii, 2024-07-29:
Hi @atan2005 

What is your module's current firmware? You can check by sending the AT command `ATI`. 

Also as @MiG-41  mentioned, can you try again with the default configuration? It should work out of the box.

atan2005, 2024-07-29:
`ATI`
Quectel
RG500Q-EA
Revision: RG500QEAAAR11A07M4G_ACY
OK

As per my replies to @MiG-41, it still didn't work for me.

xhikarishii, 2024-07-29:
@atan2005 now I see where the problem is :)

You are using an outdated firmware of the module, you can update your module using `qfirehose`, the firmware is also on my repositories

xhikarishii, 2024-07-29:
Here [quectel-rm500q-ea](https://github.com/xhikarishii/quectel-rm500q-ea)

atan2005, 2024-07-29:
My device is a Telstra branded Arcadyan AW1000 which I got back in 03-Jun-2024 and it was running Telstra firmware version 0.06.01r, hardware version: XCI35AX44Z-TA before I replace it with OpenWrt snapshots dated 30-Jun-2024, then QWrt, then HikariWrt.

Could you please confirm if it is fine to proceed to update my Quectel modem with the firmware [RG500QEAAAR13A01M4G_01.200.01.200.zip](https://github.com/xhikarishii/quectel-rm500q-ea/blob/main/RG500QEAAAR13A01M4G_01.200.01.200.zip) that you have provided if you consider my modem is still on updated Telstra custom firmware? Thanks.

xhikarishii, 2024-07-29:
@atan2005 should be fine, I came from the same module firmware version too when I got mine

atan2005, 2024-07-30:
As a side note, the extracted files can be scp over from Windows pc, this time the files have been extracted into a USB flash drive designated d: formatted with either exFAT, FAT or NTFS partition being the 3 most commonly available partition types on Windows pc. 

````
C:\>dir d:
 Volume in drive D has no label.
 Volume Serial Number is 240D-1A8E

 Directory of d:\

13/06/2023  02:59 AM             1,740 md5.txt
13/06/2023  02:59 AM             5,710 contents.xml
30/07/2024  09:36 AM    <DIR>          update
12/07/2023  06:40 PM           915,410 Quectel_RG500Q-EA-AA_软件版本变更说明_V1301_01.200.01.200.pdf
12/07/2023  06:40 PM           852,693 Quectel_RG500Q-EA-AA_Firmware_Release_Notes_V1301_01.200.01.200.pdf
               4 File(s)      1,775,553 bytes
               1 Dir(s)   1,891,336,192 bytes free
````

Before inserting USB drive...
````
root@HikariWrt:/# ls -l /dev/sd*
ls: /dev/sd*: No such file or directory
````

After inserting USB drive...
````
root@HikariWrt:/# ls -l /dev/sd*
brw-------    1 root     root        8,   0 Jul 30 09:42 /dev/sda

root@HikariWrt:/# block info | grep "/dev/sd"
/dev/sda: UUID="240D-1A8E" TYPE="exfat"

root@HikariWrt:/# lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 001: Dev 002, If 0, Class=[unknown], Driver=option, 5000M
    |__ Port 001: Dev 002, If 1, Class=[unknown], Driver=option, 5000M
    |__ Port 001: Dev 002, If 2, Class=[unknown], Driver=option, 5000M
    |__ Port 001: Dev 002, If 3, Class=[unknown], Driver=option, 5000M
    |__ Port 001: Dev 002, If 4, Class=[unknown], Driver=qmi_wwan_q, 5000M
````

It detected the usb drive with exfat filesystem but it did not detect the first and only partition `/dev/sda1` with the extracted files in it. From the information I have found, your release lacks these packages kmod-fs-exfat, kmod-fs-vfat, kmod-fs-ntfs3 to support the 3 most common filesystems on Windows. When I tried to install the relevant packages, it complained about the versions mismatch as below:-

[image]

[image]

[image]

root@HikariWrt:/# uname -r
6.6.37

Your firmware version of 6.6.37 versus current available version: 6.6.43 dated 30-Jul-2024 and going forward only newer snapshot versions will be available. Could you please advise how do I resolve the versions mismatch issue and also going forward if more software are needed?

MiG-41, 2024-07-30:
Please look out during mode update , if it will stuck on some reason , also whole router will not boot: https://forum.openwrt.org/t/arcadyan-aw1000-telstra-5g-smart-modem-support-ipq807x/122307/115

Just to be warned. 

Easiest way is to use WinSCP and use SCP protocol to transfer files ,like here: https://www.youtube.com/watch?v=uyI2_B024Nk

xhikarishii, 2024-07-30:
@atan2005 let's focus on the current issue, and create one on separate thread so we don't get confused here :)

xhikarishii, 2024-08-05:
Hi @atan2005 did that worked for you? Since I haven't received any updates for a week I'll close this issue.

closed 2024-08-05 by xhikarishii

atan2005, 2024-08-15:
@xhikarishii Sorry I have been way and have no physical access to the router at home. Could you please reopen this issue?

I haven't pursued the firmware update and have some concern due to @MiG-41 post about this OpenWrt account `stich86` having his modem (he didn't specify his device detail then) is stuck into endless reboot after performing a modem firmware update, he mentioned:

> There was an update, it went fine using fire but the the RG500 start to reboot itself between 9008 and fastboot.

Am back today and I have researched further into all his related postings on OpenWrt in order to find out which device he was referring to! I have found he has mentioned [ZTE MF269](https://openwrt.org/toh/hwdata/zte/zte_mf269), which may have similar modem but entirely a different hardware than our [Arcadyan / Astoria AW1000, 1GB](https://openwrt.org/toh/hwdata/arcadyanastoria/arcadyanastoria_aw1000). Do you think his mishap could be due to this?

To get mine right, the first (and last) time, could you please provide me with the exact command I need to run the qfirehose, assuming I already have all the zip extracted files and folders stored under `/tmp`.