xhikarishii/openwrt-ipq #3: Quectel Cellular, QMI Cellular, ModemManager Protocols (Part 2)

atan2005, 2024-08-16:
### Describe the bug

@xhikarishii Sorry I have been away and have no physical access to the router from remote. Could you please reopen this issue?

I didn't pursue with the firmware update as I had concern due to @MiG-41 post about this OpenWrt account `stich86` having his modem (he didn't specify his device detail then) stuck on endless reboot after performing a modem firmware update, he mentioned:

> There was an update, it went fine using fire but the the RG500 start to reboot itself between 9008 and fastboot.

Am back today and I have researched further into all his related postings on OpenWrt in order to find out what device he was referring to! He owned a [ZTE MF269](https://openwrt.org/toh/hwdata/zte/zte_mf269) in his [Qualcommax NSS Build](https://forum.openwrt.org/t/qualcommax-nss-build/148529/2618) posting, which may have similar modem but an entirely different hardware than [Arcadyan / Astoria AW1000](https://openwrt.org/toh/hwdata/arcadyanastoria/arcadyanastoria_aw1000) in his [Arcadyan AW1000 (Telstra 5G Smart Modem) support-IPQ807x](https://forum.openwrt.org/t/arcadyan-aw1000-telstra-5g-smart-modem-support-ipq807x/122307/115) posting. Could this be his mishap?

I have watched the [Quectel - How to upgrade module thru Linux QFirehose](https://forums.quectel.com/t/how-to-upgrade-module-thru-linux-qfirehose/15556) and to use the `-f` option.

```
root@HikariWrt:/tmp/quectel# ls -l
-rw-r--r--    1 root     root        852693 Aug 16 14:13 Quectel_RG500Q-EA-AA_Firmware_Release_Notes_V1301_01.200.01.200.pdf
-rw-r--r--    1 root     root        915410 Aug 16 14:13 Quectel_RG500Q-EA-AA_软件版本变更说明_V1301_01.200.01.200.pdf
-rw-r--r--    1 root     root          5710 Aug 16 14:13 contents.xml
-rw-r--r--    1 root     root          1740 Aug 16 14:13 md5.txt
drwxrwxrwx    3 root     root           480 Aug 16 14:13 update

root@HikariWrt:/tmp/quectel# qfirehose
[000.000]: Version: QFirehose_Linux_Android_V1.4.9
[000.000]: Builded: Jul 10 2024 02:43:07
[000.000]: Upgrade Quectel's modules with Qualcomm's firehose protocol.
[000.000]: Usage: qfirehose [options...]
[000.000]:     -f [package_dir]               Upgrade package directory path
[000.000]:     -p [/dev/ttyUSBx]              Diagnose port, will auto-detect if not specified
[000.000]:     -s [/sys/bus/usb/devices/xx]   When multiple modules exist on the board, use -s specify which module you want to upgrade
[000.000]:     -e                             Erase All Before Download (will Erase calibration data, careful to USE)
[000.000]:     -l [dir_name]                  Sync log into a file(will create qfirehose_timestamp.log)
[000.000]:     -u [usbmon_log]                Catch usbmon log and save to file (need debugfs and usbmon driver)
[000.000]:     -n                             Skip MD5 check

root@HikariWrt:/tmp/quectel# qfirehose -f /tmp/quectel
[000.000]: Version: QFirehose_Linux_Android_V1.4.9
[000.000]: Builded: Jul 10 2024 02:43:07
[000.000]: Find md5 check file </tmp/quectel/md5.txt>
[000.000]: md5 checking: /tmp/quectel/contents.xml pass
[000.002]: md5 checking: /tmp/quectel/update/abl.elf pass
[000.003]: md5 checking: /tmp/quectel/update/aop.mbn pass
[000.003]: md5 checking: /tmp/quectel/update/apdp.mbn pass
[000.003]: md5 checking: /tmp/quectel/update/devcfg.mbn pass
[000.003]: md5 checking: /tmp/quectel/update/firehose/partition_complete_p4K_b256K.mbn pass
[000.003]: md5 checking: /tmp/quectel/update/firehose/patch_p4K_b256K.xml pass
[000.005]: md5 checking: /tmp/quectel/update/firehose/prog_firehose_sdx55.mbn pass
[000.005]: md5 checking: /tmp/quectel/update/firehose/rawprogram_nand_p4K_b256K_update.xml pass
[000.006]: md5 checking: /tmp/quectel/update/hyp.mbn pass
[000.006]: md5 checking: /tmp/quectel/update/ipa_fws.elf pass
[000.006]: md5 checking: /tmp/quectel/update/multi_image.mbn pass
[000.515]: md5 checking: /tmp/quectel/update/NON-HLOS.ubi pass
[000.540]: md5 checking: /tmp/quectel/update/oemdata.ubi pass
[000.540]: md5 checking: /tmp/quectel/update/partition.mbn pass
[000.540]: md5 checking: /tmp/quectel/update/partition_nand.xml pass
[000.544]: md5 checking: /tmp/quectel/update/sbl1.mbn pass
[000.607]: md5 checking: /tmp/quectel/update/sdxprairie-boot.img pass
[000.698]: md5 checking: /tmp/quectel/update/sdxprairie-recoveryfs.ubi pass
[001.224]: md5 checking: /tmp/quectel/update/sdxprairie-sysfs.ubi pass
[001.224]: md5 checking: /tmp/quectel/update/sec.dat pass
[001.227]: md5 checking: /tmp/quectel/update/tools.fv pass
[001.232]: md5 checking: /tmp/quectel/update/tz.mbn pass
[001.240]: md5 checking: /tmp/quectel/update/uefi.elf pass
[001.264]: md5 checking: /tmp/quectel/update/usrdata.ubi pass
[001.265]: md5 checking: /tmp/quectel/update/xbl_cfg.elf pass
[001.265]: Totals checking 26 files md5 value, 0 file fail!
[001.266]: [1] /sys/bus/usb/devices/4-1 2c7c/800/414
[001.266]: P: /dev/bus/usb/004/002 idVendor=2c7c idProduct=0800
[001.266]: C: /dev/bus/usb/004/002 bNumInterfaces: 5
[001.266]: I: If#= 0 Alt= 0 #EPs= 2 Cls=ff Sub=ff Prot=30
[001.266]: E: Ad=81 Atr=02 MxPS= 1024 Ivl=0ms
[001.266]: E: Ad=01 Atr=02 MxPS= 1024 Ivl=0ms
[... 692 lines cut ...]
OK

### OpenWrt version

r26991-3ac5a2b54f

### OpenWrt release

v1.1.0b

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Arcadyan AW1000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MiG-41, 2024-08-16:
So again , default configuration is for QMI , and your modem is in MBIM... (+QCFG: "usbnet",2 ) . Set it to QMI , cycle poweroff/on ( power not reset) , and then show all commands: 
AT+QCFG="usbnet"
AT+CGDCONT?
AT+CGCONTRDP
AT+QNWPREFCFG="mode_pref"
AT+QNWPREFCFG="lte_band"
AT+QNWPREFCFG="nsa_nr5g_band"
AT+QNWPREFCFG="nr5g_band"
AT+QMBNCFG="Select"
AT+QMBNCFG="AutoSel"
AT+QMBNCFG="List"

atan2005, 2024-08-17:
@MiG-41 When only running QMI Cellular protocol and having MBN deactivated, see as below.

`AT+QCFG="usbnet"`
+QCFG: "usbnet",0
OK

`AT+QCFG=?`
+QCFG: "rrc",(0-5)
+QCFG: "hsdpacat",(6,8,10-24)
+QCFG: "hsupacat",(5,6)
+QCFG: "pdp/duplicatechk",(0,1)
+QCFG: "risignaltype",("respective","physical")
+QCFG: "lte/bandprior",(1-43),(1-43),(1-43)
+QCFG: "volte_disable",(0,1)
+QCFG: "diversity/config",(4,6),(1-4),(0)
+QCFG: "div_test_mode",(0,1)
+QCFG: "usbspeed",("20","30")
+QCFG: "data_interface",(0,1),(0,1)
+QCFG: "pcie/mode",(0,1)
+QCFG: "pcie_mbim",(0,1)
+QCFG: "sms_control",(0,1),(0,1)
+QCFG: "call_control",(0,1),(0,1)
+QCFG: "usb/maxpower",(0-900)
+QCFG: "efratctl",(0,1)
+QCFG: "netmaskset",(0,1)[,<netmask>]
+QCFG: "mmwave",ant_chip,ant_type
+QCFG: "gatewayset",(0,1)[,<gateway>]
+QCFG: "clat",(0,1),(0,1),<prefix>,(0,32,40,48,56,64,96),<fqdn>,(0,1),(0,1,2,4,8),(0,1),(0,1),(0,1,2),(0,1,2)
+QCFG: "usage/apmem"
+QCFG: "enable_gea1"[,(0,1)]
+QCFG: "dhcppktfltr",(0,1)
+QCFG: "rollback",(0,1)
OK

`AT+CGDCONT?`
+CGDCONT: 1,"IPV4V6","","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0,,,,,,,,,"",,,,0
OK

`AT+CGDCONT=?`
+CGDCONT: (1-42),"IP",,,(0-3),(0-4),(0-1),(0-1),,,,,,,,,(0-1),,(0-1),,,(0-1)
+CGDCONT: (1-42),"PPP",,,(0-3),(0-4),(0-1),(0-1),,,,,,,,,(0-1),,(0-1),,,(0-1)
+CGDCONT: (1-42),"IPV6",,,(0-3),(0-4),(0-1),(0-1),,,,,,,,,(0-1),,(0-1),,,(0-1)
+CGDCONT: (1-42),"IPV4V6",,,(0-3),(0-4),(0-1),(0-1),,,,,,,,,(0-1),,(0-1),,,(0-1)
OK

`AT+CGCONTRDP`
+CGCONTRDP: 1,5,telstra.internet,10.100.101.187,32.1.128.4.20.0.13.255.10.107.57.115.87.3.192.15, 254.128.0.0.0.0.0.0.0.0.0.0.0.0.0.1,10.3.41.226 32.1.128.4.44.66.9.28.0.0.0.0.0.0.0.1,10.3.24.11 32.1.128.4.45.67.9.0.0.0.0.0.0.0.0.1
OK

`AT+CGCONTRDP=?`
+CGCONTRDP: ( 1 )
OK

`AT+QNWPREFCFG="mode_pref"`
+QNWPREFCFG: "mode_pref",AUTO
OK

`AT+QNWPREFCFG="lte_band"`
+QNWPREFCFG: "lte_band",1:3:5:7:8:18:19:20:26:28:32:34:38:39:40:41:42:43
OK

`AT+QNWPREFCFG="nsa_nr5g_band"`
+QNWPREFCFG: "nsa_nr5g_band",1:3:5:7:8:20:28:38:40:41:77:78:79
OK

`AT+QNWPREFCFG="nr5g_band"`
+QNWPREFCFG: "nr5g_band",1:3:5:7:8:20:28:38:40:41:77:78:79
OK

`AT+QMBNCFG="Select"`
OK

`AT+QMBNCFG="AutoSel"`
+QMBNCFG: "AutoSel",0
OK

`AT+QMBNCFG="List"`
+QMBNCFG: "List",0,0,0,"Telstra_Australia_Commercial",0x0A010F00,202201111
+QMBNCFG: "List",1,0,0,"Dito_Commercial",0x0A01FD09,202101151
+QMBNCFG: "List",2,0,0,"Telia_Sweden",0x0A012400,202101151
+QMBNCFG: "List",3,0,0,"TIM_Italy_Commercial",0x0A012B00,202101151
+QMBNCFG: "List",4,0,0,"France-Commercial-Orange",0x0A010B21,202101151
+QMBNCFG: "List",5,0,0,"Commercial-DT-VOLTE",0x0A011F1F,202101151
+QMBNCFG: "List",6,0,0,"Germany-VoLTE-Vodafone",0x0A010449,202101151
+QMBNCFG: "List",7,0,0,"UK-VoLTE-Vodafone",0x0A010426,202101151
+QMBNCFG: "List",8,0,0,"Commercial-EE",0x0A01220B,202101151
+QMBNCFG: "List",10,0,0,"Commercial-LGU",0x0A012608,202101151
+QMBNCFG: "List",11,0,0,"Commercial-KT",0x0A01280B,202101151
+QMBNCFG: "List",12,0,0,"Commercial-SKT",0x0A01270A,202101151
+QMBNCFG: "List",13,0,0,"Commercial-Reliance",0x0A011B0C,202210211
+QMBNCFG: "List",16,0,0,"Commercial-DCM",0x0A010D0D,202101151
+QMBNCFG: "List",18,0,0,"VoLTE_OPNMKT_CT",0x0A0113E0,202101151
+QMBNCFG: "List",19,0,0,"Volte_OpenMkt-Commercial-CMCC",0x0A012010,202101151
OK

[image]

[image]

[image]

MiG-41, 2024-08-18:
You havn't set a proper APN and/or PDP in Network/Interfaces/5g .
It also shows AT+CGDCONT? command.

atan2005, 2024-08-19:
@MiG-41 In your first replay above you have asked me to set to QMI. The APN `telstra.internet` with PDP `IPv4/IPv6` were set correctly on the `5g` Network Interface when only running `QMI Cellular` protocol and somehow that AT command did not display the correct config.

My 28 day Telstra sim card expired yesterday and today I have to buy a new 28 day Amaysim sim card (using Optus network with APN `yesinternet`). The overall experience is still the same like in my [Part 1](https://github.com/xhikarishii/openwrt-ipq/issues/2) posting back in July, still not working. See below.

Now, when only running `Quectel Cellular` protocol.

`AT+QCFG="usbnet"`
+QCFG: "usbnet",0
OK

`AT+CGDCONT?`
+CGDCONT: 1,"IPV4V6","yesinternet","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0,,,,,,,,,"",,,,0
OK

`AT+CGCONTRDP`
OK

`AT+QNWPREFCFG="mode_pref"`
+QNWPREFCFG: "mode_pref",AUTO
OK

`AT+QNWPREFCFG="lte_band"`
+QNWPREFCFG: "lte_band",1:3:5:7:8:18:19:20:26:28:32:34:38:39:40:41:42:43
OK

`AT+QNWPREFCFG="nsa_nr5g_band"`
+QNWPREFCFG: "nsa_nr5g_band",1:3:5:7:8:20:28:38:40:41:77:78:79
OK

`AT+QNWPREFCFG="nr5g_band"`
+QNWPREFCFG: "nr5g_band",1:3:5:7:8:20:28:38:40:41:77:78:79
OK

`AT+QMBNCFG="Select"`
OK

`AT+QMBNCFG="AutoSel"`
+QMBNCFG: "AutoSel",0
OK

`AT+QMBNCFG="List"`
+QMBNCFG: "List",0,0,0,"Telstra_Australia_Commercial",0x0A010F00,202201111
+QMBNCFG: "List",1,0,0,"Dito_Commercial",0x0A01FD09,202101151
+QMBNCFG: "List",2,0,0,"Telia_Sweden",0x0A012400,202101151
+QMBNCFG: "List",3,0,0,"TIM_Italy_Commercial",0x0A012B00,202101151
+QMBNCFG: "List",4,0,0,"France-Commercial-Orange",0x0A010B21,202101151
+QMBNCFG: "List",5,0,0,"Commercial-DT-VOLTE",0x0A011F1F,202101151
+QMBNCFG: "List",6,0,0,"Germany-VoLTE-Vodafone",0x0A010449,202101151
+QMBNCFG: "List",7,0,0,"UK-VoLTE-Vodafone",0x0A010426,202101151
+QMBNCFG: "List",8,0,0,"Commercial-EE",0x0A01220B,202101151
+QMBNCFG: "List",9,0,0,"Optus_Australia_Commercial",0x0A014400,202101151
+QMBNCFG: "List",10,0,0,"Commercial-LGU",0x0A012608,202101151
+QMBNCFG: "List",11,0,0,"Commercial-KT",0x0A01280B,202101151
+QMBNCFG: "List",12,0,0,"Commercial-SKT",0x0A01270A,202101151
+QMBNCFG: "List",14,0,0,"Commercial-SBM",0x0A011C0B,202101151
+QMBNCFG: "List",15,0,0,"Commercial-KDDI",0x0A010709,202101151
+QMBNCFG: "List",16,0,0,"Commercial-DCM",0x0A010D0D,202101151
+QMBNCFG: "List",17,0,0,"VoLTE-CU",0x0A011561,202101151
+QMBNCFG: "List",18,0,0,"VoLTE_OPNMKT_CT",0x0A0113E0,202101151
+QMBNCFG: "List",19,0,0,"Volte_OpenMkt-Commercial-CMCC",0x0A012010,202101151
+QMBNCFG: "List",20,0,0,"ROW_Commercial",0x0A010809,202201191
OK

[image]

[image]

[image]

[image]

It is the same outcome like above even after multiple power on/off!

MiG-41, 2024-08-19:
There is no connection to tower at all... 
What shows AT+CFUN? command ?
if different then 1 , please send AT+CFUN=1

atan2005, 2024-08-19:
@MiG-41 I think you have hit the jackpot!

`AT+CFUN?`
+CFUN: 0
OK

`AT+CFUN=1`
OK
+C5GREG: 2

After a device reboot, the value went back to 0 and I have to set it again to for `5G` to connect!!

This may also explain why temporarily switching to `QMI Cellular` and then back to `Quectel Cellular` would make `5G` to connect! (Or add `ModemManager` in parallel for connection that can survive a reboot!)

atan2005, 2024-08-19:
@MiG-41 Does your device always have `AT+CFUN=1` when `Quectel Cellular` protocol running even after reboot?

MiG-41, 2024-08-19:
I don't have to do AT+CFUN=1 manually , and i read AT+CFUN? always 1 , so yes ,always have 1.

And witch parameters are changing ? QIND shows some logs , CREG shows information about tower , so if not connected is nothing ... 

And if you refer , that on 3ginfo ,when you set AT command you recive additional trashes (or sometimes nothin ) , it is just a "non polished" plugin ....

atan2005, 2024-08-19:
@MiG-41 I see. Please see below, no luck.

`AT&F0`
OK

`AT+CFUN?`
+CFUN: 1
OK

After reboot...

`AT+CFUN?`
+CFUN: 0
OK

xhikarishii, 2024-08-19:
Removing the bug label as this is not related to my build, also please do write on Quectel forums for module-related issues.

closed 2024-08-19 by xhikarishii

MiG-41, 2024-08-19:
Maybe it is the case of command AT+CREG , and have to be set as well  to 1 or 2 ...
Maybe after setting CFUN and CREG it is needed to "save settings" via AT&W ... 
Try it....

atan2005, 2024-08-19:
@MiG-41 Tried as below, same result.

Upon boot up...

`AT+CREG?`
OK+CREG: 0,0
OK

`AT+CFUN?`
+CFUN: 0
OK

`AT+CFUN=1`
OK
+C5GREG: 2

`AT+CREG?`
+CREG: 2,1,"CB25","1361632",7
OK

`AT&W`
OK

After reboot...

`AT+CREG?`
OK+CREG: 0,0
OK

`AT+CFUN?`
+CFUN: 0
OK

MiG-41, 2024-08-19:
set also AT+CREG=1 ,AT+CFUN=1 and then AT&W

atan2005, 2024-08-19:
@MiG-41 Have tried, but no luck.

Under Quectel forum, have found something similar and may be related...

[AT Commands not saving settings](https://forums.quectel.com/t/at-commands-not-saving-settings/9821)

[BG96 not saving +QCFG=“nwscanmode” after reset](https://forums.quectel.com/t/bg96-not-saving-qcfg-nwscanmode-after-reset/2864)