xhikarishii/openwrt-ipq #4: PBR and mwan3 support

apa-u, 2024-08-16:
Hello, Hikari! How are you? I’d like to ask if you’ll be able to support PBR on your version of openwrt for AW1000? I’m asking this as I tried installing _pbr_ and _luci-app-pbr_ and unfortunately on _luci-app-pbr_ there’s an incompatibility with _kmod-ipt-core_and requires _kernel version 6.6.45_ same goes with _mwan3_. Thank you

xhikarishii, 2024-08-19:
Hi @apa-u , I am looking into this as the current quectel-cm has compatibility issues with both PBR and MWAN3. Will include it on the next release if I manage to fix the issue :)

apa-u, 2024-08-24:
Thanks a bunch @xhikarishii! I really appreciate your work on the build. If there’s anything I can do to support you, please let me know. Also, if it’s not too much trouble, would you be able to add PCI locking? If not, no worries at all!

xhikarishii, 2024-08-28:
> Thanks a bunch @xhikarishii! I really appreciate your work on the build. If there’s anything I can do to support you, please let me know. Also, if it’s not too much trouble, would you be able to add PCI locking? If not, no worries at all!

This is something we are working on currently ;)

xhikarishii, 2024-09-25:
Kernel support for PBR and MWAN3 will be added on next release, closing this :)

closed 2024-09-25 by xhikarishii