xhikarishii/openwrt-ipq #8: Tsping for cake autorate compiling

apa-u, 2024-09-27:
Hello Hikari, how are you?

I'm currently testing your new build and I want to thank you for including PBR and mwan3. Hopefully, you'll be able to integrate QMI soon as well.

I also have a question: I'm planning to compile _tsping_ for _Cake-Autorate_ and I'm curious about which SDK version you're using to build this firmware. I'm trying to optimize my cake-autorate settings on my device as it greatly helps with my latency and buffer bloat. 

I'm still new to this, but I’d like to take it on as a weekend project. I'm following a guide to compile it for my AW1000. Here's the link Im following: [link](https://github.com/Lochnair/tsping)

xhikarishii, 2024-09-27:
Hi @apa-u the current repo is a fork of Qosmio's NSS builds which uses the snapshot SDK

apa-u, 2024-09-27:
Thank you hikari! I was able to compile mine. 🫡😊

xhikarishii, 2024-09-27:
You're always welcome, don't hesitate to reach again if there's anything. 👍