xhikarishii/openwrt-ipq #10: Connectivity issue

telemach1337, 2024-10-09:
### Describe the bug

I encounter a problem with auto connecting to network. All settings are correct but firmware doesn't see modem.

What helps is switching protocol in Interfaces-5G to QMI and then back to Quectell 
or going to terminal and running 

ifdown 5g
uqmi -d /dev/cdc-wdm0 --set-device-operating-mode online
ifup 5g

but having that in rc.local doesn't help so after power loss I need to run whole procedure once again

BTW : Can you make some setting to store default APN configuration ? because each time I switch protocol I have to fill it again

### OpenWrt version

hikariwrt-v1.1.2b-091924

### OpenWrt release

hikariwrt-v1.1.2b-091924

### OpenWrt target/subtarget

hikariwrt-v1.1.2b-091924

### Device

AW1000

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MiG-41, 2024-10-09:
Maybe in rc.local before that comands you need to make some sleep ( start with sleep 60 ). 

It is just workaround , but maybe somhow helps.

telemach1337, 2024-10-13:
Is it normal that being very close to BTS the AW1000 is aggregating only B1+B7+n78 and transfers are not bigger than 300 ?
any other router is aggregating B1+B7+B20+B8+n78 and transfers are higher than 400-500

xhikarishii, 2024-10-14:
Hi, this is a known issue for quectel modules, you can try sending `AT+CFUN=1` during boot time.

xhikarishii, 2024-10-14:
Also, make sure to switch back to `qmi/rmnet` mode using the AT command `AT+QCFG="usbnet",0`, the default configurations made to the firmware should work out of the box.

closed 2024-12-13 by xhikarishii