xhikarishii/openwrt-ipq #11: Delays on future releases - APK migration

xhikarishii, 2024-11-26:
Hello guys, as you may know, openwrt is no longer hosting snapshot IPK packages and have moved to APK packages, I am still yet to decide whether I should move too or host my own IPK packages for the meantime as both takes time. 

Any suggestion would help, or if you want to help with hosting and cost, you can reach me personally.

ChamodyaChiran, 2024-11-27:
I think moving to APK packages is a good idea. Hosting your own IPK packages could work as a temporary solution, but it might lead to compatibility and update issues in the future. I'm also trying to install Passwall, but it has already been updated to APK packages and no longer has IPK packages available.Your OpenWRT build is absolutely amazing.

xhikarishii, 2024-11-28:
Yeah, same thoughts, as NSS is also moving towards APK, but there are still old packages, and I doubt that they will be having APK releases soon. But let's see.

ChamodyaChiran, 2024-12-02:
Released today.
https://mirror-03.infra.openwrt.org/releases/24.10.0-rc1/

xhikarishii, 2024-12-03:
Oh finally, looks we are having an owrt full support :)

closed 2024-12-13 by xhikarishii