xhikarishii/openwrt-ipq #13: ROOter Support

xhikarishii, 2025-03-02:
Hi guys, as you know, moving forward my releases will be using ROOter along with NSS. 

If find any bugs and issues, please feel free to report them here.

matrock47, 2025-03-13:
Hi Hikari, Can i know where can i get R00ter repo and which repo you are using? I try to use R00ter repo unfortunately there is no target device for aw1000. is there any method to compile openwrt-ipq with r00ter dialer?

Thank you in advance.

xhikarishii, 2025-03-13:
Hi @matrock47, I used the openwrt-ipq as base for NSS and minimal ROOter packages for the modem management, all from the repo.

amorim, 2025-04-09:
Hello @xhikarishii, first thanks for this build, it works great.

There is one minor issue I'm experiencing though, I don't know if it's a bug or some misconfiguration on my side. Whenever the wwan interface reconnects, the wan interface (wired ethernet) goes down on mwan3. After this happens, I can only ping the gateway IP for the wan interface, can't ping any other IP (meanwhile the wwan/wan1 interface continues to work fine, and mwan3 switches traffic to it).

Restarting the wan interface fixes the issue (until the next wwan reconnect), so for now I just added a script that automatically restarts the wan interface if it goes down. Do you have any idea of what I can do to fix this issue?

xhikarishii, 2025-04-11:
Hi @amorim, I know the instability issues with mwan3, as I have experienced the same with my fiber + 5g setup. For now, you can use the `internet-detector` for restarting the interface of the disconnecting WANs

closed 2025-06-24 by xhikarishii