xhikarishii/openwrt-ipq #14: READ: Bootloop on some devices

xhikarishii, 2025-03-04:
There have been reports of some AW1000 users experiencing bootloops after updating to version `1.2.0b-rooter`. I'm already looking into the issue and will post a fix soon.

For the meantime, if you have upgraded to the said version, you can downgrade to 1.1.3b by following these steps. 

1. Remove the sim card from the device.
2. Boot normally in to the device and flash the known stable version.
3. Wait for the device to reboot and retry setting up your sim card.

xhikarishii, 2025-03-04:
Update: I already found the issue, and a fix is underway.

tconnard, 2025-03-04:
Ha, I was about to write a post on the AW1000 NSS build thread about how to debug this.
Here is the last few messages before thing reboots for what it is worth.
`Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.163411] NSS core 1 signal COREDUMP COMPLETE 4000
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.163450] ffffffc0791f5340: Starting NSS-FW logbuffer dump for core 1
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.167458] ffffffc0791f5340: Warn: trap[813]: Trap on CHIP ID 00050002
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.173789] ffffffc0791f5340: Warn: trap[620]: Trapped: TRAP_TD(00000040) DCAPT(3C000080)
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.180405] ffffffc0791f5340: Warn: trap[645]: Trapped: Thread: 6, reason: 00000800, PC: 40824544, previous PC: 40824540
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.188739] ffffffc0791f5340: Warn: trap[594]: A0_3: 40C36100 3F00A0A8 00000000 0000000C
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.199665] ffffffc0791f5340: Warn: trap[594]: A4_7: 40823AF4 4081413C 3F02F00C 3F006F68
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.207742] ffffffc0791f5340: Warn: trap[599]: D0_3: 00000000 3F00A0A8 00000000 00000000
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.215811] ffffffc0791f5340: Warn: trap[599]: D4_7: 40C35800 00000004 00000400 00000000
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.223885] ffffffc0791f5340: Warn: trap[599]: D8_11: 0000000C 0000000F 00000000 00000000
Sun Mar  2 22:38:14 2025 kern.alert kernel: [  116.231953] ffffffc0791f5340: Warn: trap[599]: D12_15: 00000000 00000000 00000000 00000000`

xhikarishii, 2025-03-04:
@tconnard yep, this happens when the script initiates the interface for the `qmi_wwan_q` seems that the RMNET patches broke after the last commits on NSS. Digging deeper now.

xhikarishii, 2025-03-04:
@tconnard feel free to share your insights, more help is always welcome 😄

tconnard, 2025-03-04:
I am not running the buged firmware at the moment so I am going from memory but I did have an issue with the atinout command in the system logs, don't know how relevant it is. Pretty sure it was line 70 in quectel.sh causing it.

xhikarishii, 2025-03-04:
@tconnard the thing is, I'm also running the current firmware with `atinout` that is missing but I haven't experienced bootloops. Seems to be there's something at the kernel that is behaving strangely on some devices. But I do have one tester that had the same errors from the logs that is coming from the kernel.

tconnard, 2025-03-04:
Yeah this smells more like a qmi firmware thing.
The other notable thing about my unit is that I got it directly from Telstra with original firmware, not secondary market with qwrt.

xhikarishii, 2025-03-04:
> Telstra with original firmware

This confirms my theory that something in the kernel is not being updated properly, I also posted the issue on the NSS thread for further discussion.

xhikarishii, 2025-03-05:
@mouhmmad you can also stop the 5G modem to connect automatically by setting `Modem > Connection Profile > Advanced > Stop After Detection > Yes`

xhikarishii, 2025-03-05:
Then maybe you can use RNDIS for the meantime while the fix for RMNET support is on it's way

fathiwll, 2025-03-08:
Please try the new version.

xhikarishii, 2025-03-10:
New release with the hotfix is up https://github.com/xhikarishii/openwrt-ipq/releases/tag/v1.2.1b-ROOter

xhikarishii, 2025-03-10:
Let me know if it's fixed on your end @tconnard , @mouhmmad

fathiwll, 2025-03-10:
Fixed the problem of restarting by itself

xhikarishii, 2025-03-10:
Glad to hear it's fixed on your side too @fathiwll! Will confirm with the others too before closing this thread 🙂

jocca1997, 2025-03-10:
Hello, i updated from qwrt to ur last version, but cant connect on 4G using MBIM protocol? Is there any fix? Thanks

xhikarishii, 2025-03-10:
@jocca1997 this build doesn't use MBIM, you can read the instructions on the release page on how to switch to QMI/RMNET

fathiwll, 2025-03-11:
There is a problem in this version when putting the network on 4G only Keeps catching on the net 3G Although that current cellular mode Set to  4G only

xhikarishii, 2025-03-11:
@fathiwll okay I'll look into this, can you also put the issue here so I can gather them? [ROOter Support](https://github.com/xhikarishii/openwrt-ipq/issues/13)

xhikarishii, 2025-03-11:
@fathiwll  For the meantime, you can try using the band locking feature and uncheck those you are not using

fathiwll, 2025-03-11:
It is not possible now because the network is back 4G When it is out of service, I can put a picture or video. Because when it is cut off 4G Automatically the network changes to 3G I want to disconnect from the network when it is disconnected 3G To make room for another tower to call that has 4G Until the connection is restored to the first tower

fathiwll, 2025-03-11:
Is it possible to increase the Wi-Fi hotspot feature?So I can contact the network wifi Other

xhikarishii, 2025-03-11:
@fathiwll I'm not really sure what you mean by hotspot feature, but if you mean to bypass hotspot restrictions by your operators, you can tweak the TTL values on the connection profile settings.

fathiwll, 2025-03-11:
I am talking about, for example, a person from the neighbors who gives me a code wifi I connect to his network via modem

fathiwll, 2025-03-11:
[image]

fathiwll, 2025-03-11:
ok Next time you can put a light version and a full version for those who want more features in the modem Thanks for the updates.

xhikarishii, 2025-03-12:
Will close this bug thread, as three more of users that experienced the bootloop issues from the last release is also fixed for them.

closed 2025-03-12 by xhikarishii