xhikarishii/openwrt-ipq #15: How to remove "MWAN" while keeping the "miscellaneous"(Band / Cell lock)

lgmadurange, 2025-03-13:
I installed the latest version, v1.2.1b-ROOter, but it takes too long to connect to the internet. Even when it does connect, the connection is unstable and frequently disconnects.

To fix this, I removed MWAN3, and after that, the connection worked perfectly. However, removing MWAN3 also removed the entire modem interface, including the modem configuration submenu.

How can I remove MWAN3 while keeping the Miscellaneous section (Band/Cell lock)?

Note: Version v1.1.3b works fine, but it doesn’t have the Miscellaneous (Band/Cell lock) feature. I would also be fine if I could add that feature to v1.1.3b instead.

xhikarishii, 2025-03-14:
Hi @lgmadurange 

Removing MWAN3 should not affect the interface as it's being created by ROOter itself, for the meantime maybe you can try just disabling MWAN3 at `System > Startup > MWAN3 >`

lgmadurange, 2025-03-14:
Hi @xhikarishii ,

Thank you for your response.

I tried that, but it had no effect. The connection still takes too long and remains unstable. Is there any possibility of adding the "Miscellaneous (Band/Cell lock) feature" to version v1.1.3b ?

xhikarishii, 2025-03-14:
Unfortunately, it's not possible since it's a ROOter feature.

xhikarishii, 2025-05-01:
I am considering removing MWAN3 from future releases as it's currently causing an issue with NSS. I'll look into a possible solution, or else I will just add support for it.